skuep/AIOC #90: Unable to adjust TX audio levels from AllStarLink

adamdhardy, 2024-10-14:
I have flashed 1.3 to the AIOC and changed the ID to emulate the CM108. I am using AllStarLink 3.0.4.

PTT and COS seem to be working properly. 
The TX audio is clear, but a little low, especially when ID’ing.

I have found that I cannot adjust the TX audio level from the SimpleUSB Interface Tune CLI, (options 3 & 4) It seems to have no effect at all.  Changing the ‘idamplitude’ in the  rpt.conf file also has no effect. 

I have tried 2 different AIOC with no change. If I switch to a CM108 fob the settings work as expected again.

Have I missed something is the setup? Does the AIOC have some kind of AGC that I’m fighting against?

WROG208, 2024-10-14:
You use the volume knob on the radio. Simpleusb doesn't do anything.

skuep, 2024-10-16:
The AIOC should have enough oomph to drive a microphone input. Are you sure you have set the TX volume correctly?

skuep, 2025-01-04:
I will close this issue due to inactivity. Feel free to reopen if you still have issues. In that case it would be interesting to know what Radio you are using?

closed 2025-01-04 by skuep