skuep/AIOC #91: Python script: No module named 'hid'

W1BTR, 2024-10-18:
When attempting to run the python script on a fresh linux installation I get:
```
sudo python3 aioc.py
Traceback (most recent call last):
  File "aioc.py", line 2, in <module>
    import hid
ModuleNotFoundError: No module named 'hid'
```

skuep, 2024-10-18:
I think you need to install "hid" and not "hidapi" (I know it's stupid how many trash is now on pypi). Can you remove the hidapi and install hid?

W1BTR, 2024-10-18:
Got it, now I get this error, which I think another error reported something similar in the past on #19 

```
Traceback (most recent call last):
  File "aioc.py", line 2, in <module>
    import hid
  File "/usr/local/lib/python3.8/dist-packages/hid/__init__.py", line 125, in <module>
    hidapi.hid_get_input_report.argtypes = [ctypes.c_void_p, ctypes.c_char_p, ctypes.c_size_t]
  File "/usr/lib/python3.8/ctypes/__init__.py", line 386, in __getattr__
    func = self.__getitem__(name)
  File "/usr/lib/python3.8/ctypes/__init__.py", line 391, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: /lib/x86_64-linux-gnu/libhidapi-hidraw.so.0: undefined symbol: hid_get_input_report
```

Perhaps it's related to the fact Im running this on WSL, although the port is passed through to the container so it _shouldnt_ change anything.

skuep, 2024-10-19:
Looks to be a version issue with the libraries in WSL probably being quite old.
https://medium.com/devopss-hole/python-error-undefined-symbol-hid-get-input-report-c01af1ccf1f6

````
pip uninstall hid
pip install hid==1.0.4
````

W1BTR, 2024-10-19:
New error upon doing that:
```
lucas@MONEY-PIT:~$ sudo lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 1209:7388 Generic All-In-One-Cable
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

lucas@MONEY-PIT:~$ sudo python3 aioc.py
Traceback (most recent call last):
  File "aioc.py", line 70, in <module>
    aioc = hid.Device(vid=0x1209, pid=0x7388)
  File "/usr/local/lib/python3.8/dist-packages/hid/__init__.py", line 130, in __init__
    raise HIDException('unable to open device')
hid.HIDException: unable to open device
lucas@MONEY-PIT:~$
```

skuep, 2024-10-19:
Well, that's s verbose error message! 😅
No idea, unfortunately. I know of people using this under Windows (using Windows hid driver backend) and I am using it under native Linux with hidapi backend. Not sure if WSL trips about the libusb stuff or something

yamez1969, 2024-12-29:
I created a live Ubuntu USB flash drive with persistent storage. Booted the USB off my Windows PC, installed the python and pip packages, edited the python HID script and now have Virtual PTT running. Tested and working with JS8Call, LongChat, RattleGram and AndFlmsg so far.

skuep, 2025-01-04:
Those Windows issues are really hit and miss unfortunately. I can't provide support for Windows, mostly because I don't run any Windows anymore personally. These days it feels like it is a good idea to switch away from Windows if you want to play around with open source projects like the AIOC. Or at least get a Raspberry Pi running for doing these kinds of things.

I will close the issue for now, feel free to reopen if you have any further issues or information.

closed 2025-01-04 by skuep

W1BTR, 2025-01-16:
My initial issue that I opened was not for Windows. I am still unable to get this to work on **any** system, be it Linux, Windows, or WSL.

Currently I am attempting on an Ubuntu 24.04 machine. It's **not** running in WSL or anything. I get:

```
❯ sudo python3 hidaioc.py
Traceback (most recent call last):
  File "/home/max/aioc/hidaioc.py", line 70, in <module>
    aioc = hid.Device(vid=0x1209, pid=0x7388)
           ^^^^^^^^^^
AttributeError: module 'hid' has no attribute 'Device'. Did you mean: 'device'?
```

I found where it said `aioc.Device` and changed it to a lower case. I then get:

```
Traceback (most recent call last):
  File "/home/max/aioc/hidaioc.py", line 72, in <module>
    magic = Struct("<L").pack(read(aioc, Register.MAGIC))
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/max/aioc/hidaioc.py", line 53, in read
    device.send_feature_report(request)
  File "hid.pyx", line 170, in hid.device.send_feature_report
ValueError: not open
```

You dont have any install instructions for this python script. Other than python3, Im guessing python3-hid is required. Then, should the script be run as root?

I tried as sudo and without sudo, same errors.

skuep, 2025-01-16:
It may be possible, that you have `hidapi` and `hid` installed at the same time. `hidapi` does not work and unfortunately hides behind the same import name `hid`. Make sure `hidapi` is not installed either on the system (`sudo apt-get remove python3-hidapi` probably) and pip (`pip uninstall hidapi`).

There are some infos on the package page on PyPI:
https://pypi.org/project/hid/#description

W1BTR, 2025-01-16:
No luck, hidapi was not installed, only python3-hid.

yamez1969, 2025-01-17:
I can't exactly remember the steps I took to install hid and successfully excute the Auto PTT AIOC script on live Ubuntu 24.04.1 LTS USB flash drive with persistent storage. Installing hid related packages with --break-system-packages, does ring a bell. I just flashed the latest AIOC firmware and tested the script again and the Ubuntu USB is still working. I'll look to flashing another stick to recreate the hid install steps and make notes this time! Auto PTT AIOC script attached.

Side note: Auto PTT is working with DireWolf + Reticulum, been testing for a week now. I've now switched to serial PTT for better control.

[aioc-auto-PTT.zip](https://github.com/user-attachments/files/18447819/aioc-auto-PTT.zip)

Radio-X24, 2025-01-17:
Windows is still suffering with issues i think, ive tried unsuccessfully
for days but for pi/ubuntu etc the following 3 lines can help

You may or may not need to use sudo:
dfu-util: apt-get install dfu-util
HID: pip install hid
libfuse2: apt install libfuse2 libhidapi-hidraw0 libhidapi-libusb0

skuep, 2025-01-17:
I don't have windows, so I need some feedback on what works and what doesn't.

At this point these python scripts are just small snippets that I threw together quickly. There are plans to make better aioc configuration tools, but for that I need to know what hid library to use. I think I tried the hidapi library before but there were some issues if I remember correctly. If anyone knows the best solution for cross platform hid communication, let me know.

W1BTR, 2025-01-18:
> Windows is still suffering with issues i think, ive tried unsuccessfully
> for days but for pi/ubuntu etc the following 3 lines can help
>
> You may or may not need to use sudo:
> dfu-util: apt-get install [...]

What command do you use to run it? I use `python3 script-name.py` (also pro-tip, dont name it hid.py lol)

j0ju, 2025-02-16:
Alternativly you can use a python3 virtualenv.
I Installed a fresh RaspiOS 64 bit bookworm, then I prepared a virtual python3 environment so that the systems python is not harmed.
1. apt install libhidapi-hidraw0 libhidapi-libusb0

```
# prepearing the virtual environment
[root@cm4:/root]# mkdir aoic-py
[root@cm4:/root]# cd aoic-py
[root@cm4:/root/aoic-py]# virtualenv .
created virtual environment CPython3.11.2.final.0-64 in 899ms
  creator CPython3Posix(dest=/root/aoic-py, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/root/.local/share/virtualenv)
    added seed packages: pip==23.0.1, setuptools==66.1.1, wheel==0.38.4
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
[root@cm4:/root/aoic-py]# . bin/activate
(aoic-py) [root@cm4:/root/aoic-py]# bin/pip install -U pip
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pip in ./lib/python3.11/site-packages (23.0.1)
Collecting pip
  Downloading https://www.piwheels.org/simple/pip/pip-25.0.1-py3-none-any.whl (1.8 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.8/1.8 MB 3.2 MB/s eta 0:00:00
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 23.0.1
    Uninstalling pip-23.0.1:
      Successfully uninstalled pip-23.0.1
Successfully installed pip-25.0.1

# installing requirements
(aoic-py) [root@cm4:/root/aoic-py]# bin/pip install hid
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting hid
  Downloading https://www.piwheels.org/simple/hid/hid-1.0.7-py3-none-any.whl (4.9 kB)
Installing collected packages: hid
Successfully installed hid-1.0.7

# copying the script
(aoic-py) [root@cm4:/root/aoic-py]# cp ~/aoic.py .

(aoic-py) [root@cm4:/root/aoic-py]# bin/python3 aoic.py
Manufacturer: AIOC
Product: All-In-One-Cable
Serial No: c7c1ca7b
Magic: b'AIOC'
Current PTT1 Source: 1028
Current PTT2 Source: 8
Current CM108 Button 1 (VolUP) Source: 131072
Current CM108 Button 2 (VolDN) Source: 16777216
Current CM108 Button 3 (PlbMute) Source: 0
Current CM108 Button 4 (RecMute) Source: 0
```

The `requirements.txt` would look like: ;)
```
hid
```

skuep, 2025-02-16:
That's really weird that you need the 32 bit environment. Gotta say, I am not impressed with the problems I see on the hid module... 

Regarding the PTT issue, can you do a continuity check on the TRS pins? I had it happen sometimes that after soldering one of the connections through the TRS connector is broken. What did help is to exert some force on the solder tabs while they solidify, so that the connection is under slight spring tension.

guywire, 2025-02-17:
Yeah, there's probably other options but it looks like this issue has been open for 4 months and this solution is reproducible. A venv in python should work too, but I couldn't get it to work.

I resoldered my connectors just for fun and it still doesn't work. Is there a component that's most likely to fail in the ptt path? I'll pull up the desing in kicad and see if I can figure that out when I get a chance.

I also submitted a change to the new version of aprsdroid to add the pid and vid of AIOC as it seems to be not in their list. I think if that works it will allow PTT via RTS. I think I did the hex to binary conversion correctly...

W1BTR, 2025-06-06:
> virtualenv .

That's definitely not a comprehensive list - virtualenv is not a command on a fresh raspberry pi 64 bit install.

skuep, 2025-06-09:
Can you post your script?

W1BTR, 2025-06-09:
I have not written a script, Im looking for guidance on how to use yours - or is that script not the thing we should be modifying?

Here is the script you posted on the hardware v1.2 issue #93 - not sure if I should use that one or the one with the 1.30 firmware release.

```
import sys
import hid
from struct import Struct
from enum import IntEnum, IntFlag

class Register(IntEnum):
    MAGIC = 0x00
    USBID = 0x08
    AIOC_IOMUX0 = 0x24
    AIOC_IOMUX1 = 0x25
    CM108_IOMUX0 = 0x44
    CM108_IOMUX1 = 0x45
    CM108_IOMUX2 = 0x46
    CM108_IOMUX3 = 0x47
    SERIAL_CTRL = 0x60
    SERIAL_IOMUX0 = 0x64
    SERIAL_IOMUX1 = 0x65
    SERIAL_IOMUX2 = 0x66
    SERIAL_IOMUX3 = 0x67
    AUDIO_RX = 0x72
    AUDIO_TX = 0x78
    VPTT_LVLCTRL = 0x82
    VPTT_TIMCTRL = 0x84
    VCOS_LVLCTRL = 0x92
    VCOS_TIMCTRL = 0x94

class Command(IntFlag):
    NONE = 0x00
    WRITESTROBE = 0x01
    DEFAULTS = 0x10
    RECALL = 0x40
    STORE = 0x80

class PTTSource(IntFlag):
    NONE = 0x00000000
    CM108GPIO1 = 0x00000001
    CM108GPIO2 = 0x00000002
    CM108GPIO3 = 0x00000004
    CM108GPIO4 = 0x00000008
    SERIALDTR = 0x00000100
    SERIALRTS = 0x00000200
    SERIALDTRNRTS = 0x00000400
    SERIALNDTRRTS = 0x00000800
    VPTT = 0x00001000

class CM108ButtonSource(IntFlag):
    NONE = 0x00000000
    IN1 =  0x00010000
    IN2 =  0x00020000
    VCOS = 0x01000000

class RXGain(IntEnum):
    RXGAIN1X = 0x00000000
    RXGAIN2X = 0x00010000
    RXGAIN4X = 0x00020000
    RXGAIN8X = 0x00030000
    RXGAIN16X = 0x00040000

class TXBoost(IntEnum):
   TXBOOSTOFF = 0x00000000
   TXBOOSTON = 0x00000100

def read(device, address):
    # Set address and read
    request = Struct('<BBBL').pack(0, Command.NONE, address, 0x00000000)
    device.send_feature_report(request)
    data = device.get_feature_report(0, 7)
    _, _, _, value = Struct('<BBBL').unpack(data)
    return value

def write(device, address, value):
    data = Struct('<BBBL').pack(0, Command.WRITESTROBE, address, value)
    device.send_feature_report(data)

def cmd(device, cmd):
    data = Struct('<BBBL').pack(0, cmd, 0x00, 0x00000000)
    device.send_feature_report(data)

def dump(device):
    for r in Register:
        print(f'Reg. {r.value:02x}: {read(device, r.value):08x}')

aioc = hid.Device(vid=0x1209, pid=0x7388)

magic = Struct("<L").pack(read(aioc, Register.MAGIC))

if (magic != b'AIOC'):
    print(f'Unexpected magic: {magic}')
    sys.exit(-1)

print(f'Manufacturer: {aioc.manufacturer}')
print(f'Product: {aioc.product}')
print(f'Serial No: {aioc.serial}')
print(f'Magic: {magic}')

if False:
    # Load the hardware defaults
    print(f'Loading Defaults...')
    cmd(aioc, Command.DEFAULTS)

if False:
    # Dump all known registers
    dump(aioc)

if False:
    # Set RX settings
    rxgain = RXGain.RXGAIN4X
    print(f'Setting Audio RX gain to {str(rxgain)}')
    write(aioc, Register.AUDIO_RX, rxgain)

if False:
    # Set TX settings
    txboost = TXBoost.TXBOOSTON
    print(f'Setting Audio TX boost to {str(txboost)}')
    write(aioc, Register.AUDIO_TX, txboost)

if False:
    # Store settings into flash
    print(f'Storing...')
    cmd(aioc, Command.STORE)
```

W1BTR, 2025-06-09:
Awesome! So at this point I have the script running on that pi, and I have a decent idea of how to configure it. I'm going to leave this issue open out of hopes that some people figure out how to get it going in a virtual python environment / on 64 bit systems consistently and reproducably (ideally on wsl as well, as a lot of customers only run Windows). Ty for the help as always, skuep!

W1BTR, 2025-06-09:
One thing I see in the example 1.3-r2 script is the following:

```
if False:
    # Swap PTT1/PTT2
    ptt1_source, ptt2_source = ptt2_source, ptt1_source
    print(f'Setting PTT1 Source to {str(ptt1_source)}')
    write(aioc, Register.AIOC_IOMUX0, ptt1_source)
    print(f'Setting PTT2 Source to {str(ptt2_source)}')
    write(aioc, Register.AIOC_IOMUX1, ptt2_source)

    print(f'Now PTT1 Source: {str(PTTSource(read(aioc, Register.AIOC_IOMUX0)))}')
    print(f'Now PTT2 Source: {str(PTTSource(read(aioc, Register.AIOC_IOMUX1)))}')
```
This swaps PTT1 and PTT2 if I'm understanding it correctly. Should I remove that chunk of code to keep ptt1 and ptt2 the expected values?

skuep, 2025-06-10:
> This swaps PTT1 and PTT2 if I'm understanding it correctly. Should I remove that chunk of code to keep ptt1 and ptt2 the expected values? 

You can, but you don't need to

> Or does anything that starts with `if False:` not get run, in which case I'd need to set that to `True` for the VPTT?section

Yep, that's exactly how it works 👍 i.e. by default the script doesn't modify anything 😀 good luck!

W1BTR, 2025-06-13:
Got it, so for the autoptt on ptt1 it's:

```
if True:
    # Set to AutoPTT on PTT1
    print(f'Setting PTT1 Source to {str(PTTSource.VPTT)}')
    write(aioc, Register.AIOC_IOMUX0, PTTSource.VPTT)
    print(f'Now PTT1 Source: {str(PTTSource(read(aioc, Register.AIOC_IOMUX0)))}')
    print(f'Now PTT2 Source: {str(PTTSource(read(aioc, Register.AIOC_IOMUX1)))}')
```

Sorry to triple confirm lol

skuep, 2025-06-14:
Looks find to me, from what I can tell 😀

skuep, 2025-06-14:
If you would like to persist the changes until next power cycle, then yes. Theoretically you can play around and if something goes wonky, just replug the AIOC. Once you are sure everything works, you can store it to the device

skuep, 2025-10-06:
It looks a bit like installing the hid package system wide is a problem on the raspberry pi due to system package conflicts? Hmm, I wish hid access was easier in python