skuep/AIOC #95: Amphenol USB-C Port Replacement Request for JLCPCB Service

stardado, 2024-10-29:
Could you please change the USB-C port? The Amphenol 12401832E40 is very difficult to obtain. To use the JLCPCB service, it would be very helpful if this could be changed.

Currently, the following USB-C port is available: https://www.lcsc.com/product-detail/USB-Connectors_SHOU-HAN-TYPE-C-24P-QT_C2681555.html

I would be very grateful.

WROG208, 2024-10-29:
You can outsource the parts and have them sent directly to jlcpcb. The amphenol i got from a source in the US and sent to China took about 3 weeks. Changing the USB has been asked in the past but is the best to work with the AIOC from what i remembered reading. There was a specific reason. Search the past ticket's

stardado, 2024-10-29:
I understand your point, it wouldn't be feasible for Germany as the device would quickly cost over $40-50. Amphenol is also hard to source here, and shipping to China simply wouldn't be economically worthwhile. But Simon will probably know that. I hope we can offer a second variant instead. Unfortunately, I'm not familiar enough with KiCad to make the change myself.

WROG208, 2024-10-29:
I tried kicad to add my name to the silkscreen so i knew if anyone i sold an AIOC and asked for help or send it back i knew it was one of mine but kicad kept saying that I could only make the changes on the file that was made in AutoCAD before it was converted to whatever format it was at the moment.
Like i mentioned before search the tickets there has been a couple asking about the same thing can't remember if there was a solution for it.

stardado, 2024-10-31:
Thank you for your feedback. Perhaps Simon can do this directly, it should actually be very quick if you are familiar with KiCad. However, I have decided to use EasyEDA, another tool is too much for me.

rhgndf, 2024-10-31:
There is a PR #68 that solves this.

skuep, 2024-10-31:
I want to keep the design free of "chinese" parts (that are jlcpcb or lcsc specific) in case anyone wants to manufacturer this in a local shop that uses a "regular" distributor. 
You can use the BOM tool in your jlcpcb account to pre-order stuff directly at jlcpcb. It takes a few days but then the parts are in your local stock and you can use them in the assembly. I did this myself last weekend and it took until tuesday.

stardado, 2024-10-31:
That was a very important tip, I can understand that you don't want to install ‘China’ parts here. I found the service ‘Global Sourcing Parts’, so I was able to order the required quantity. Now let's wait and see, thanks for your help.

rhgndf, 2024-10-31:
That footprint is also more available from a "regular" distributor, since there are at least two parts with the same footprint
USB4105-GF-A: https://mm.digikey.com/Volume0/opasdata/d220001/medias/docus/5492/USB4105.pdf
CU3216S2SDLR013-NH: https://octopart.com/datasheet/cu3216s2sdlr013-nh-cvilux-147957006

CU3216SASDLR010: non stocked, so not counted

skuep, 2024-10-31:
Thanks. It's really a lot of work to find alternative sources for a given footprint (that's also one of the reasons for not using something else). 

The first two parts are widely available on the western market.
https://octopart.com/de/search?q=10155435-00011LF&currency=USD&specs=0
https://octopart.com/de/search?q=USB4105-GF-A&currency=USD&specs=0

I might have a look at the alternative amphenol part.

rhgndf, 2024-11-01:
Usually sorting by price would show the parts with the most common footprints. Using mouser, on the first page I found a few more:

USB4215-03-A
UJ20-C-H-C-4-SMT-TR
GSB1C411111DS1HR
217179-0001

skuep, 2024-11-03:
See, it's not that easy. 217179-0001 and USB4215-03-A are not compatible mechanically. They have slightly different spacing on the shell pads.

I think I am going to switch to Amphenol 10155435-00011LF for hardware Rev. 1.2 
It will be compatible to the connector proposed by #PR68 and the GCT parts you mentioned

rhgndf, 2024-11-03:
Ah yes, thanks for noticing it! I have updated the comment to reflect that.

skuep, 2025-01-04:
V1.2 is coming along well and will be using an easier to source USB port. I will thus close this issue for now.
See https://github.com/skuep/AIOC/pull/93

closed 2025-01-04 by skuep