skuep/AIOC #98: Is there a mistake in the schematic STM32F302CBT6 is used

uguraltinsoy, 2024-12-07:
The STM32F302CBT6 pins in the schematic you shared here do not match, can you help?

https://github.com/skuep/AIOC/blob/master/doc/images/k1-aioc-schematic-rev1.0.pdf

https://www.lcsc.com/datasheet/lcsc_datasheet_2304140030_STMicroelectronics-STM32F302CBT6_C94046.pdf
[image]

skuep, 2024-12-08:
It's not a "mistake" per se. It's just that I started out with the STM32F373 in mind, because it has better ADCs, but it was hard to source. So I decided on the (somewhat compatible) STM32F302, which is more widespread.
The symbol is from the STM32F373 and mostly matches the STM32F302, but some pins are connected such that both MCUs could potentially be used.

uguraltinsoy, 2024-12-08:
Do you suggest I do it by pin names and not by pin numbers?

skuep, 2024-12-08:
Do *what* exactly? For the pins that we are using, both pin names and pin numbers (should) match your correct symbol.

uguraltinsoy, 2024-12-08:
I can merge it into your project if you check when I'm done

uguraltinsoy, 2024-12-08:
Updated Schematic I hope there is no mistake, I made some changes that will reduce the price, if the general schematic is correct, I wanted to provide convenience for amateur friends who will do it.

[image]

skuep, 2024-12-09:
Should be fine, as said the pins we are using should have the same pin name **and** number as you have now. Or did you see any deviation?

uguraltinsoy, 2024-12-09:
PF6 and PF7 “STM32F302CBT6” are not available

skuep, 2024-12-09:
Yeah, but we are not using those here

uguraltinsoy, 2024-12-09:
Then there seems to be no problem, but could you please take a quick look at it?

skuep, 2025-01-04:
Sorry I have very limited time to check the schematic. One thing to note though is, that I don't recommend using the same regulator for 3.3V analog and digital due to the confirmed issue of USB 1kHz SOF signal bleed thru into the audio signals. You can't nicely filter such a low frequency (except with a second linear regulator).

I am going to close this issue now. Feel free to reopen if you have specific questions

closed 2025-01-04 by skuep