skuep/AIOC #104: PTT remains engage when using AIOC with ASL3 on a Pi 3

pepepr, 2025-01-22:
I got 2 devices and both have the same issue. At this point I do not know if there is a defect on the devices, or what particular setting is incorrect in the configuration.

Here is where I am:
 - AIOC has both plugs properly attached to the device. To make it easier, I've used wires from the plates of the AIOC to the plug (No need to worry about perfect alignment etc.)
 - The AOIC is recognized by the Pi OS.
 - The latest firmware has been installed on the AIOC (1.3)
 - The configuration on rpt.conf has been made to the specifications I found. This includes using invertedusb, etc.
 - This node is going to be use as a bridge using DV Switch.
 - I've attempted to use this device with ASL 3 as a stand alone, ASL 3 with DV Switch, and HamVoip as a stand alone. All scenarios present the same issue.
 - Without the AIOC present, the node works as intended in all configurations and I can make connections to and from it.

The issue:
The main problem is that the PTT remains engaged and it is not released.
The only way to disengage the PTT is to unplug the device or restart the Pi.
On initial load of the Pi, when all is connected, there is no engagement of the PTT suggesting that the initial communication between the 2 devices on load is clean.

There is something that is causing the PTT to remain engaged.
I would appreciate any specific configuration that we would need to set, any exiting posts that deal with the same issue or any particular firmware or software version that we would need to use.
I've done plenty of google searches and reading trying to come up with a solution but nothing has worked, so I've done my homework.

Any help would be greatly appreciated. Thank you.

Radio-X24, 2025-01-22:
Sounds like it's locking up, out of curiosity what antenna are you using

pepepr, 2025-01-22:
> Sounds like it's locking up, out of curiosity what antenna are you using
> […](#)

The radio rubber duck.

Radio-X24, 2025-01-22:
That's probably it then, however wait for developers input

pepepr, 2025-01-22:
> When does the PTT start to be engaged?

The problem is happening when ASL 3 is transmitting, it will not release the PTT. If I key a radio and the node receives, it is fine. As soon as ASL transmit, it get stuck.

daffster, 2025-02-23:
For what its worth, I've experienced this issue before.
I tried a few things to resolve it until I was successful. You might want to try too
1) shorter USB cable to the Pi (mine is 15cm long now)
2) putting a few clip on ferrite beads on the USB cable, nearest the AIOC.
3) switching from a Baofeng uv-5r to a Quangsheng UV-K5
4) lowering the TX power to 250mW

As mentioned by @Radio-X24 this is only a problem with the rubber duck. If I connect up my external dualband antenna, this doesn't happen.

I believe this is stray RF getting into the PTT circuit on the AIOC (via the USB cable?) and holding it open.

schuyler, 2025-04-22:
Hi all -- I was having a similar issue with a Baofeng UV-5R using the factory "rubber duck" antenna. I am running Direwolf on both MacOS and Linux, with the same problem on both operating systems.

**Reducing TX power from 5W to 1W on the Baofeng cleared up all of the issues** I was having with the AIOC locking up and keeping TX on.

WROG208, 2025-05-24:
I eliminated this by adding pigtails instead of soldering the 3.5 and 2.5 jacks directly to the AIOC. Someone had problems only once, and it was solved with a ferrite. I started with the AIOC at the start (almost) and never liked the soldering of the jacks because it was a bit of a PITA, and I broke plenty of jacks getting the legs straight, and even lost a couple of AIOC by damaging the soldering pads. Since then, all my AIOC have pigtails instead.
The one problem I was having was that after letting go of the PTT, it would take a second and a half to disengage, but as soon as the 1.5 sec were over, the interference would engage the PTT again for another 1.5 sec, making the total tail 3 sec or more. With the pigtails and ferrite that were eliminated completely.

closed 2025-09-25 by skuep