skuep/AIOC #106: Station speaker not mutting as AIOC connecting

Vegaukr, 2025-02-17:
After assembling and uploading firmware to AIOC everything was looking good. I connect it to my Baofeng and its internal speaker is not mute. Comunication with station is not established. Please any ideas? What in schematic of station and AIOC responsable for recognition of connected to kenwood jack external device.

Vegaukr, 2025-02-24:
Baofeng radios typically don’t rely on mechanical switches inside the jack itself to detect an external connection. Instead, they likely use a resistance-based detection method.
  I tryed to connect unsoldered connector to both jacks and station is not muted. So it might be some  resistance. The radio mutes the internal speaker when it detects an external load on the speaker output (3.5mm jack).

Vegaukr, 2025-04-02:
Some Baofeng radios expect to see ~10kΩ  resistor to GND internally on the speaker line to mute extermal speaker. I have an old Baofeng and AIOC not recognised by my baofeng. In my case if I connect an "empty" plug (without a speaker), the built-in speaker of the UV-5R remains on
It means that I have a CHIRP APRS Direwolf and other applications not working. But I have Qunsheng UV-K5 and AIOC working perfect in all aplications. I check schematic for audio input line of AOIC and think that 10 k resistor before C14 to GND can be a solution for some stations which don’t rely on mechanical switches inside the jack itself to detect an external connection.

skuep, 2025-09-26:
Interesting, I did not know that. You are the first one to have this issue actually (that I heard of at least). There are currently no plans to include a DC path to ground on the Speaker Plug on the PCB.

Interestingly enough, my UV-5R works without issues, but who knows what they changed over time. For the odd radio, that doesnt work, it might make sense to just solder a 0603 SMD resistor at the pad.

I am going to close the issue, until I see this popping up more often, but thanks for the information!