skuep/AIOC #107: JLCPCB Error

tkisason, 2025-04-10:
Hello, 

I tried fabricating a few AIOCs with the help of JLCPCB according to the steps outlined in the Readme. When i get to the BOM and CPL upload steps i get the following error from JLCPCB:

```
The below parts won't be assembled due to data missing.
H1,H2 designators don't exist in the BOM file.
D4,J2,R17,D3 designators don't exist in the CPL file.
```

Am i doing something wrong or should the BOM / CPL be updated?

Thank you for your effort and for the cool project! Can't wait to test it out. 

73 9A3CQS

Radio-X24, 2025-04-10:
https://github.com/skuep/AIOC/blob/de71b070808e92b6b467326d4d4e0f8afc9ccad7/README.md?plain=1#L67

tkisason, 2025-04-10:
Ah, RTFM solves the day, thank you! Closing this. Thank you!

closed 2025-04-10 by tkisason

skuep, 2025-04-10:
Jep 😀