skuep/AIOC #109: 5V "VCC" source for C-IV Communication?

W1BTR, 2025-05-20:
ICOM C-IV uses a single cable for RX and TX data. This is its schematic:

[image]

Is there an easy way to tap into a VCC (5V) Constant 5V source on the AIOC?

W1BTR, 2025-05-21:
I found the 3V2 pin on the STM32 (the pin one connects to the "BOOT" pin to flash firmware initially and attempted to use that.

Despite the fact that this _should_ work, it seems to totally disrupt the AIOC's serial communication and flat out wont work. Sucks because a lot of radios use CI-V. Wonder how the digirig did this.

skuep, 2025-05-22:
Interesting.. there are some protection resistors on the aioc PCB, maybe they cause an issue? 

I also noticed from a quick glance that the stm32 has half-Duplex usart support built in, which should theoretically do what you need. 

Is there any cheapish radio that I can get for testing this?

W1BTR, 2025-05-22:
It's largely for their HF radios, so usually a good chunk of change, but they do hold their value pretty well (assuming you buy used).

Heck Ill sell you the IC-706MKIIG Im testing with for a discounted price, say $390 is pretty much as low as I could go.

Anyone around you got an ICOM HF radio you could borrow?

skuep, 2025-05-24:
I see.. I really don't know anyone with such a radio, but I will have a look at the feature when I have time next.

Btw, can you try the schematic from above without the 10K resistor? And maybe a schottky diode instead of 1N4148? BAT54 or something

W1BTR, 2025-06-10:
I don't know of any schottky diodes that will switch fast enough to support CI-V, definitely none I have. I tried it without the 10K resistor and no luck so far, never receives any response. A good tool is CI-V scout, though I initially 

I'll gladly pay $50 to anyone who can get CI-V figured out with the AIOC / AIOB!