skuep/AIOC #111: Where is hardware COS on the 1.2 schematic?

W1BTR, 2025-06-12:
Hi all! I'm new with KiCAD and I'm struggling to find the pad for hardware COS on the 1.2 schematic. Could you point me in the right direction?

[image]

skuep, 2025-06-13:
If I remember correctly IN2 is used for COS by default, i.e. PB7

W1BTR, 2025-06-13:
Ah ok, I thought there was a separate COS pad on the bottom of revision 2, that just connects with PTT2?

W1BTR, 2025-06-14:
I dont believe so. Maybe yall refer to it as COR, basically the PC can detect that the radio is RXing, which is useful for interfacing with some devices or for allstarlink for example.

skuep, 2025-06-15:
PTT2 is an open drain output (i.e. can only pull low or leave hi-Z). If PTT2 is hi-Z, it can work as an input.

So in essence it's connected to OUT2 and IN2 simultaneously 😀

W1BTR, 2025-06-15:
[image]

skuep, 2025-06-15:
You can use the PTT2 pad that is used for the TRS connector 😁 Tip contact of the 3.5mm plug