skuep/AIOC #112: A python cmd line util for managing AIOC

hrafnkelle, 2025-06-15:
I created a command line utility based on python code I found in this repo and others to inspect and change register values of the AIOC. My initial motivation was to help fellow radio amateurs that needed to enable VPTT to do so more easily.

https://github.com/hrafnkelle/aioc-util

skuep, 2025-06-15:
Awesome, Looks pretty cool!
I will put a link in the readme when I have some time

Btw, there are even more registers where things can be configured 😁 they are just hidden away in the firmware and not included in the sample scripts I put out there. If you are interested I can dig them out and see if there are settings that might be useful to others

hrafnkelle, 2025-06-15:
I've been looking at the firmware too. More than happy to add more registers. Any suggestions welcome.

skuep, 2025-11-02:
@hrafnkelle Pretty cool, thanks a lot! 

I will test it if I have time, until then others are free to test this as well.