skuep/AIOC #113: Update README for  K1 AIOC Hardware Rev 1.2 wirering

mdphoto, 2025-06-19:
Hi,

Please add this information on README :  when a TRS connector is used. If not, then both pads can be used separately

I don't use TRS directly  on board, I use cable and TRS.

I make K1 AIOC Rev 1.2 with JLCPCB (It's very easy with your work, thank)

I lost more time after find : when a TRS connector is used. If not, then both pads can be used separately on [PR](https://github.com/skuep/AIOC/pull/93)

Thank for your work, AIOC work with Reticulum.

skuep, 2025-09-26:
A pinout image is now available in the README
https://github.com/skuep/AIOC?tab=readme-ov-file#how-to-use

sethpark777, 2025-10-31:
So the correct usage of that split pad is to connect to both if you are using the board normally? I'm using pigtails rather than trs so I was looking for clarification in your readme but never got the clarification I was looking for. And maybe it's lack of knowledge on my part, but I really don't fully understand the purpose of the split pad after studying the readme. I know the pad is labeled in that image but it was unclear to me what it's to be used for. Again might just be lack of knowledge on my part.

skuep, 2025-10-31:
On the TRS connector of the HTs, the pad is shared between PTT and serial data (for programming the radio mainly). So when using the TRS connector z both signals are connected together.

When used without, you just connect the signal that you actually need. If you need just the PTT, but the serial needs to go to another line, you can connect two separate cables to the two pads and they are completely separated.