skuep/AIOC #116: Please help to use JLCPCB

vonpupp, 2025-07-23:
Hi,

I am new to JLCPCB. I am following the instructions on the readme **on branch rev 1.2** as follows:

```
    Go to JLCPCB.com and upload the GERBER-k1-aioc.zip package (under kicad/k1-aioc/jlcpcb)
        Select PCB Thickness 1.2mm (that is what I recommend with the TRS connectors I used)
        You may want to select LeadFree HASL
        Select Silkscreen/Soldermask color to your liking
    Check "PCB Assembly"
        PCBA Type "Economic"
        Assembly Side "Top Side"
        Tooling Holes "Added by Customer"
        Press Confirm
        Click "Add BOM File" and upload BOM-k1-aioc.csv
        Click "Add CPL File" and upload CPL-k1-aioc.csv
        Press Next
        Look Through components, see if something is missing or problematic and press Next
        Check everything looks roughly good (rotations are already baked-in and should be correct). Save to Cart
```

Here is an image with the summary:
[image]

Questions:
- Is there any setting that needs to be changed? Does it look okay? 
- I am using **rev 1.2**. Is it functional? Can I order from this branch?
- Are the 3.5 and 2.5 Jack connectors included in the BOM file or do I need to buy them separately?
- Do you have experience with shipping method: "Global Standard Direct Line (12-16 days)" to Europe?
- Apparently it is possible to print a 3d case on the same website, do you have any experience with that?

Last but not least and probably one of the most important questions. I got a problem with one of the leds, they are in a shortage.

[image]

These are the options, but I am not sure if they can be safely replaced by some of these.

[image]

Thank you very much.

Aqueum, 2025-10-01:
@skuep hope you had a lovely holiday - you were certainly productive.  I was delighted that you managed to pull v1.2 to master.

I also found @vonpupp 's screen shots really useful when we ordered ours together in a club meeting last night.  I stupidly failed to take screenshots at the time, but have tried to recreate them this morning - possibly as an update.  It would be great to know if we got this right (the factory is closed so we still have time to change the order).  

The main difference is that they now let you choose your board material and UL Marking (is FR4 TG135 and No UL Marking fine?).  Also I'm sure we selected Yes to Deburring/Edge rounding, but it changed to 'No' post-order:

[image]

[image]

[image]

I'm also a little concerned that our part placement view isn't showing the part numbers as I think it did in a previous aborted attempt:

[image]

It would be great to have a running version of the latest optimal screenshots for JLC virgins like us - possibly linked from the README.  However I think this LED Shortage Issue can be closed now, as they are definitely back in stock.

skuep, 2025-11-02:
@JetseVerschuren Sure, no problem! 

I am going to close the issue for now, feel free to reopen if necessary

closed 2025-11-02 by skuep

vonpupp, 2025-11-16:
Thank you so much @skuep! I appreciate all the help.

macropin, 2026-01-16:
The 19-217/GHC-YR1S2/3T are out of stock again. Might be worth making the 19-217/GHC-YR1S2/6T a permanent part substitution?

skuep, 2026-01-17:
Given that the 19-217/GHC-YR1S2/3T is discontinued, it would not be a bad idea.. I will see when I get around to doing that.